DangerMonkeys/leaf #4: Firmware Updates without Arduino IDE

oxothnk423, 2024-12-15:
Feature for end-users:

Need to create & test a method for uploading new firmware that does not require Arduino IDE etc.

Possible Candidates:
- put new firmware file on SD card
- wireless over the air (OTA) updates
- simple executable for Windows/Mac computers
- possibly mobile phone app via USB-C connection

scottyob, 2024-12-31:
@oxothnk423:  Worth closing this now https://github.com/DangerMonkeys/leaf/commit/47e0c28cac5e303f6f4cf75ab1a86d8cd76920e1 has it over WiFi?  Or want to keep the SD card option on the todo list too (as a P1?)

oxothnk423, 2024-12-31:
@scottyob I think we can close for now.  Thanks again!  Also, Just tried an update solely with my phone.. did manual wifi setup, connected to "leaf" with my phone, typed in credentials for my phone hotspot, hit "save" on the wifi manager page, and as soon as it said "trying to connect" I switched on my hotspot.  Worked like a charm!  So we can even update in the field easily too if needed, with just a phone and cell signal.  Perfect!

Perhaps I'll add a note to the SDcard Mass storage issue, that if we want at that point, we can enable SD card firmware updates.  Just so we don't lose the idea.

closed 2024-12-31 by oxothnk423