DangerMonkeys/leaf #18: [Bug] Charging with audible vario makes terrible beep

scottyob, 2024-12-22:
If you set the Vario audio, then plug in the Leaf, I get a low pitched beep when it first comes up charging.  Holding down the D pad seems to initialize the device and makes the low tone go away.

oxothnk423, 2024-12-27:
I've noticed this too

I think this is an artifact of improper initialization ... either 
(a) the altitude and climb filters aren't set up properly and need time to become accurate, or 
(b) the initial pressure reading from the sensor is incorrect and several more samples are needed to "average out" the bad data point.  
 (or something like that)

You can see this when the vario is first turned on -- the alt and climb rate are fluctuating for a moment before stabilizing.

A complete fix would be:
 1) fixing the boot up initialization issues so the vario starts at the proper altitude and with [near-]zero climbrate
   AND
 2) properly disabling the speaker in charge mode so that any future errors don't create funny sounds :)

oxothnk423, 2024-12-29:
Fixed.
   Boot sequence begins with altitudes assumed to be zero, 
   then taking a pressure measurement, 
   then calculating true altitude, 
   then calculating climb rate (current altitude vs previous altitude)

   This resulted in a erroneous climb rate (comparing current altitude to zero).  

   Quick solution without rearchitecting the baro initization sequence was to set a flag 
`  bool firstClimbInitialization = true;`
 This allows us to skip the first climb rate calculation (assuming climb rate = 0), then set flag to false, and wait until the next altitude measurement so we have two true altitudes to compare for actual climbrate.

closed 2024-12-29 by oxothnk423