DangerMonkeys/leaf #89: Mass Storage setup long delays on large SD cards

oxothnk423, 2025-03-02:
Moving from 1GB to 8GB SD cards: Mass Storage setup takes ~8 seconds instead of ~1 second.  This might be acceptable when plugging into USB for charging or interfacing with computer; but it's too long of a delay when just turning on from battery power (i.e., out in the field).

At minimum: only setup mass storage when turning on by USB power, not push-button

Better: only setup mass storage when turning on by USB power AND with data connection (i.e., so we don't experience the delay when just plugging into a power source like a backup battery or solar panel or "dumb" wall charger)

If possible: find a way to accelerate mass storage setup

scottyob, 2025-03-12:
Any idea what calls are blocking on the SD card?  We may be able to move that off to it's own task and make it more async.

For what it's worth.  I've not seen it take near this long on my large SD cards

scottyob, 2025-03-13:
Unable to reproduce.  will re-open if we see it again.

closed 2025-03-13 by scottyob