DangerMonkeys/leaf #108: OTA checks or attempts sometimes fail

BenjaminPelletier, 2025-03-14:
OTA updates now seem to be unreliable -- sometimes HTTPS queries work, but sometimes they don't.  To reproduce:

1. Check out commit at v0.0.10
2. Select leaf_3_2_5_dev environment
3. Build and upload firmware
4. Turn leaf on
5. Scroll to WiFi update
6. Select Update
7. Observe success (usually), but no update because v0.0.10 is current
8. Go back one screen
9. Repeat steps 6-8 until an error is observed (usually second try for me)

Note that continuing to repeat steps 6-8 will often yield another success again eventually.

When the failure occurs, I see something like this in the Serial monitor:

```
[114293][E][NetworkClient.cpp:323] setSocketOption(): fail on 0, errno: 9, "Bad file number"
[114367][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():302]: (-32512) SSL - Memory allocation failed
[114377][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -32512
[114386][E][NetworkClient.cpp:323] setSocketOption(): fail on 0, errno: 9, "Bad file number"
```

Even on success, I see something like this in the Serial monitor:

```
[ 22281][E][NetworkClient.cpp:323] setSocketOption(): fail on 0, errno: 9, "Bad file number"
```

**Alternate reproduction method**

1. Check out commit at v0.0.9
2. Select release environment
3. Build and upload firmware
4. Turn leaf on
5. Scroll to WiFi update
6. Select Update
7. Observe success; update takes a while but succeeds -- I've tried 3 times and gotten 3 successes here
8. Turn leaf on
9. Scroll to WiFi update
10. Select Update
11. Observe failure (usually) -- if success is observed, try again

This seems to suggest this bug was introduced between 0.0.9 and 0.0.10.

scottyob, 2025-03-16:
I found using the dev build on a 3.2.5 board, it wasn't "works on the first go", but, "never works" :(  Maybe #122 is a nice dirty enough hack to close this one out???

scottyob, 2025-03-16:
Also checked with my 3.2.3 boards on prod build.  Looks like v0.0.10 with BLE support totally broke the ability to do WiFi updates all together!  This is totally not a low pri bug :(

Thanks so much for finding this so quick, @BenjaminPelletier

closed 2025-03-17 by scottyob