DangerMonkeys/leaf #125: Stopping Flight results in Crash

oxothnk423, 2025-03-16:
Even with GPS fix and SD card inserted, stopping flight results in crash.  Tested on 3.2.3hw.

scottyob, 2025-03-16:
On dev branch?  I'll try and re-create later but if you have a core dump to attach that'd be very useful!

oxothnk423, 2025-03-16:
Yes on Dev.  Is this enough (see below)?

button: CENTER state: HELD      hold count: 1
settinGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420320ba  PS      : 0x00060a30  A0      : 0x82006e6c  A1      : 0x3fcbf470  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000002  A5      : 0x3fcbf4a4  
A6      : 0x3fca9f40  A7      : 0x00000000  A8      : 0x82009e5c  A9      : 0x3fcbf440  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x3fcd3b10  A13     : 0x00000000  
A14     : 0x3fccbf60  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000015c  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x420320b7:0x3fcbf470 0x42006e69:0x3fcbf4a0 0x4200a1e1:0x3fcbf4d0 0x42017b11:0x3fcbf5b0 0x42006d59:0x3fcbf5d0 0x42012d78:0x3fcbf5f0 0x42012e23:0x3fcbf610 0x42012e70:0x3fcbf630 0x4205e32c:0x3fcbf650 0x403849a2:0x3fcbf670

ELF file SHA256: 644ef70db

Rebooting...

scottyob, 2025-03-16:
Not easily.  I was able to reproduce so have a core dump, but it's probably worth trying to see if you can get this running on your computer:  https://leafvario.com/dev-references/core-dumps/

closed 2025-03-16 by oxothnk423