DangerMonkeys/leaf #151: Occasional Bogus Climb Rate

scottyob, 2025-04-14:
We've seen a couple of instances of bogus climb rates reported, that sustains until the device is rebooted.  Sometimes it shows as -40m/s sink, sometimes a very high lift.

## Observations:
* Not based on duration of flight.  Happened to me (Scott) with Fanet device after maybe 1 hour 15 minutes of flight time
* Happened to Non-Fanet variant after only 15 minutes.
* Reboot seems to get it back into a good state.

https://github.com/user-attachments/assets/533b1921-b461-48bb-95b4-8bce8e9fe397

scottyob, 2025-09-02:
Captured a debug log event of this happening:  https://gist.github.com/scottyob/2f9a6d059ac28d4038a78c2d8abd5600

oxothnk423, 2025-12-11:
@BenjaminPelletier @scottyob 
Have we seen this recently?  There have been several improvements since the issue was created... tempted to close it..?