DangerMonkeys/leaf #220: [Crash][In-Flight] newSentence encountered in the middle of the line of text

scottyob, 2025-09-02:
Firmware version: 0.1.0-713d.dev+h3.2.6
Error: newSentence encountered in the middle of the line of text '$GPGSV,3,1,11,10,67,309,17,27,52,280,18,23,50,042,23,46,46,191,28,1*6B1*52'

Backtrace: 0x4200E3B3:0x3FCC1050 0x4200E832:0x3FCC1070 0x42010EB1:0x3FCC12C0 0x42011711:0x3FCC14A0 0x42011725:0x3FCC14C0 0x42015455:0x3FCC14E0 0x4201566B:0x3FCC1520 0x420156B5:0x3FCC1540 0x4200F7A2:0x3FCC1570 0x42018B07:0x3FCC1680 0x42018B69:0x3FCC16A0 0x42015196:0x3FCC16E0 0x4206CF14:0x3FCC1700 0x40384D82:0x3FCC1720

```
[info] Decoding 28 unique addresses...
0x4200e3b3: getBacktrace(char*, unsigned int) at /homes-local/scott/Documents/leaf/src/vario/diagnostics/fatal_error.cpp:83
0x3fcc1050: ?? ??:0
0x4200e832: fatalError(char const*, ...) at /homes-local/scott/Documents/leaf/src/vario/diagnostics/fatal_error.cpp:189
0x3fcc1070: ?? ??:0
0x42010eb1: LeafGPS::on_receive(GpsMessage const&) at /homes-local/scott/Documents/leaf/src/vario/instruments/gps.cpp:127 (discriminator 1)
0x3fcc12c0: ?? ??:0
0x42011711: bool etl::message_router<LeafGPS, GpsMessage>::receive_message_type<GpsMessage>(etl::imessage const&) at /homes-local/scott/Documents/leaf/.pio/libdeps/leaf_3_2_6_dev/Embedded Template Library/include/etl/message_router.h:519
0x3fcc14a0: ?? ??:0
0x42011725: etl::message_router<LeafGPS, GpsMessage>::receive(etl::imessage const&) at /homes-local/scott/Documents/leaf/.pio/libdeps/leaf_3_2_6_dev/Embedded Template Library/include/etl/message_router.h:450
0x3fcc14c0: ?? ??:0
0x42015455: etl::imessage_bus::receive(unsigned char, etl::imessage const&) at /homes-local/scott/Documents/leaf/.pio/libdeps/leaf_3_2_6_dev/Embedded Template Library/include/etl/message_bus.h:177
0x3fcc14e0: ?? ??:0
0x4201566b: etl::imessage_bus::receive(etl::imessage const&) at /homes-local/scott/Documents/leaf/.pio/libdeps/leaf_3_2_6_dev/Embedded Template Library/include/etl/message_bus.h:149
 (inlined by) MessageBus<(unsigned char)11>::Locked::receive(etl::imessage const&) at /homes-local/scott/Documents/leaf/src/vario/dispatch/message_bus.h:47
0x3fcc1520: ?? ??:0
0x420156b5: MessageBus<(unsigned char)11>::receive(etl::imessage const&) at /homes-local/scott/Documents/leaf/src/vario/dispatch/message_bus.h:65
0x3fcc1540: ?? ??:0
0x4200f7a2: LC86G::readLine() at /homes-local/scott/Documents/leaf/src/vario/hardware/lc86g.cpp:94 (discriminator 4)
0x3fcc1570: ?? ??:0
0x42018b07: TaskManager::updateWhileOn() at /homes-local/scott/Documents/leaf/src/vario/taskman.cpp:229
0x3fcc1680: ?? ??:0
0x42018b69: TaskManager::update() at /homes-local/scott/Documents/leaf/src/vario/taskman.cpp:141
0x3fcc16a0: ?? ??:0
0x42015196: loop() at /homes-local/scott/Documents/leaf/src/vario/main.cpp:117
0x3fcc16e0: ?? ??:0
0x4206cf14: loopTask(void*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74
0x3fcc1700: ?? ??:0
0x40384d82: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
0x3fcc1720: ?? ??:0
 *  Terminal will be reused by tasks, press any key to close it. 
```

BenjaminPelletier, 2025-09-02:
This error is triggered when there is inconsistent GPS data.  NMEA sentences are supposed to be terminated with line breaks, so a line of text is the primitive unit of data collected by leaf before processing.  During processing, the parser reports that the NMEA sentence is finished before we reach the line break.  This means the sentence is invalid, and we can confirm this by seeing that the sentence in question has two checksums.

My guess for what happened is that the system was interrupted from processing just after the last character of the NMEA sentence arrived in the buffer, and then didn't start back up until near the end of the next sentence from the GPS module.  So, the line break and entire first part of the next (and maybe more) sentences are missing.

What we do about this depends on what assumption we want to make about the validity of an NMEA sentence that isn't terminated with a line break as required by the NMEA standard.  If we assume omitting the CRLF is ok, then we can accept the sentence found mid-line and discard the rest of the line.  If we require more strict adherence to the NMEA standard, we can discard the entire line as invalid.

scottyob, 2025-09-02:
I vote for assuming it's corrupt, chucking it away on an exception, and assuming a good update will follow.