DangerMonkeys/leaf #223: [Performance] SD Card writes can back up runloops

scottyob, 2025-09-02:
* Created some telemetry for measuring IO time in the runloop:  https://github.com/scottyob/leaf/tree/file_io_telemetry

Findings:

 I added some telemetry to figure out how long some operations are taking.

Looking at the temp/humidity runtime, not much variance:

```
[scott@sob-desktop buslogs]$ {   echo "p0,p1,p5,p10,p50,p75,p90,p95,p99,p100";   grep -iE "^a" BusLog_2080-01-05_1659-2.log     | datamash -t,         min 4 perc:1 4 perc:5 4 perc:10 4 perc:50 4         perc:75 4 perc:90 4 perc:95 4 perc:99 4 perc:100 4; }
p0,p1,p5,p10,p50,p75,p90,p95,p99,p100
224,225,225,225,227,228,229,229,230,244
```

Looking at the FileWrite operations, a lot more:
```
[scott@sob-desktop buslogs]$ {   echo "p0,p1,p5,p10,p50,p75,p90,p95,p99,p100";   grep -iE "^#F" BusLog_2080-01-05_1659-2.log     | datamash -H -t,         min 2 perc:1 2 perc:5 2 perc:10 2 perc:50 2         perc:75 2 perc:90 2 perc:95 2 perc:99 2 perc:100 2; }
p0,p1,p5,p10,p50,p75,p90,p95,p99,p100
min(FileWriteTimeTakenUs(us)),perc:1(FileWriteTimeTakenUs(us)),perc:5(FileWriteTimeTakenUs(us)),perc:10(FileWriteTimeTakenUs(us)),perc:50(FileWriteTimeTakenUs(us)),perc:75(FileWriteTimeTakenUs(us)),perc:90(FileWriteTimeTakenUs(us)),perc:95(FileWriteTimeTakenUs(us)),perc:99(FileWriteTimeTakenUs(us)),perc:100(FileWriteTimeTakenUs(us))
22,25,25,26,149,155,394,417,515,939882
[scott@sob-desktop buslogs]$ 
```

The almost 1 second backup, was in writing the kalman record to the SD card:

```
M275187,A,0.976807,-0.0637207,0.192383,Q,0.54735,-0.321952,0.691105
#F275189,441
#275215,baro mb*100,101248
#F275216,150
P275215,101248
#F275216,27
#275238,kalman,6.52884845,-0.03414665,-0.03046155
#F276178,939882
M275237,A,0.976318,-0.0690918,0.19165,Q,0.547392,-0.321948,0.691093
#F276178,396
#276181,baro mb*100,101241
#F276181,146
P276180,101241
#F276181,34
```

To prevent the timing backlogs, we should move all File IO writes to their own thread.

[image]

scottyob, 2025-09-03:
Might be worth checking out some of the queues ETL provides:  https://www.etlcpp.com/concurrent_queues.html

Considering for the duration of my test, the P99 was 515us, I'm thinking it very rare that these multi-ms blocking events occur I'm thinking we would just accept the bus log as a lossy format, and, if we have to skip a few writes, perhaps we just log a counter on how many records were skipped?

I've not noticed it "backing up" on my igc logged flights over the past few months that has a different throughput, so think if we end up undersizing the buffer and skipping too many records to be useful, a move from ASCII to a binary file format for this log would be a huge, huge reduction in bytes written.

BenjaminPelletier, 2025-09-04:
> I'm thinking we would just accept the bus log as a lossy format

One of the key purposes of the bus log is to investigate behavior, and one of the most interesting places to investigate behavior is when things might back up.  This is exactly the time we wouldn't want to drop log entries.  Also, there are many debugging activities where the presence or absence of a log entry is the key piece of information -- if log entries might not exist for other reasons, this debugging approach is effectively impossible, or at least potentially very frustrating.  Also, the bus log is not the only thing we write to the SD card as mentioned above -- it would not be ok to drop GPS track or logbook SD card writes, yet we still want to immunize ourselves from long SD card writes for this data as well.  So, I don't think we would want to accept any observable write loss.

> a move from ASCII to a binary file format for this log would be a huge, huge reduction in bytes written

A big reduction in bytes written could easily be achieved by turning off [what are effectively debug messages](https://github.com/DangerMonkeys/leaf/blob/main/src/vario/logging/telemetry.h) -- KALMAN, GPS, and BARO duplicate data from the actual sensors and/or simply print computed state.  These three message types accounted for 41% of the bytes in the 8/29 bus logs.  The three remaining messages accounting for nearly all the remaining bytes are the motion message (M) at 27% of bytes, the GPS message (G) at 20% of bytes, and the pressure message (P) at 11% of bytes.

The G message's main payload is effectively already in binary -- it's reporting the raw bytes received from the GPS module and those aren't going to get smaller unless we applied some kind of compression.  The G message has three bytes of overhead (message type, comma, \n) plus timestamp per message.  8/29 timestamps averaged 6.5 bytes per timestamp.  In binary, presumably the timestamp would be a 4-byte unsigned long, so on net I think a binary format would save 3.5 bytes/G message which would reduce the bytes used for G messages by about 5%.

The M message would presumably be binary-coded with 1 byte for message type, 4 bytes for timestamp, 1 byte for validity flags, and 6*4 bytes for 6 float values = 30 bytes, versus 70 bytes on average from the 8/29 data.

The P message would presumably be binary-coded with 1 byte for message type, 4 bytes for timestamp, and 4 bytes for value = 9 bytes, versus 15 bytes on average from the 8/29 data.

With these three effects (G, M, P messages), I estimate we would reduce our byte usage by about 36% relative to today's text for the high-volume messages by switching to binary.  That's not tiny, but I don't think I would really consider it huge, huge.  And, with binary, we lose the ability to perform human-readability sanity checks (scroll to a particular place in the log and generally understand what's happening without writing code, be able to tell if things are reasonable at a glance), reduce error tolerance (\n is no longer a value reserved for termination of messages, for instance), and perform any kind of examination of the log without writing non-trivial code.

For our later reference, here's what I counted in the 8/29 logs:

| Message type | Fraction of log | Message count | Bytes per message |
|---|---|---|---|
| M | 26.5% | 382276 | 70.3 |
| baro mb*100 | 19.9% | 748160 | 27.0 |
| G | 19.8% | 302797 | 66.2 |
| kalman | 19.3% | 382276 | 51.1 |
| P | 11.1% | 748160 | 15.0 |
| gps | 2.2% | 39994 | 56.8 |
| fanet_rx | 0.6% | 7357 | 87.3 |
| A | 0.6% | 19998 | 28.5 |
| V | 0.0% | 9 | 22.1 |

> I've not noticed it "backing up" on my igc logged flights over the past few months that has a different throughput

Presumably the back up frequency is probably close to (IGC throughput) / (full-log throughput), right?  So, there are probably ~10x fewer back ups with 10x less throughput.

> Might be worth checking out some of the queues ETL provides

I would expect we would see best results by writing to a single large block of memory space rather than have each write operation be a queue element -- this would allow many write operations to be flushed to the SD card at once, which is also a more efficient way to perform the SD writes.  I don't think I'm seeing an ETL tool that does this in that queue list; perhaps they have something else that could accomplish this?